WorldWideTelescope/wwt-windows-client #7: Europa is missing

erinjohn, 2015-06-11:
In 3D Solar System mode, Europa is not visible 
[image]

doctorspaceman, 2015-06-11:
Checked this on my end (pointing to new Azure servers for testing) and it was ok.  Maybe Ron fixed it in the mean time or it was some propagation/caching issue.  But it is working now and I've confirmed this with Erin.

astrojonathan, 2015-06-12:
It is possible that it was eclipsed at the time of viewing..

erinjohn, 2015-06-12:
I turned all shadows off in the layer manager, so I don't think that was it.

Erin Johnson, WorldWide Telescope Ambassadors <http://wwtambassadors.org>

astrojonathan, 2015-06-13:
Try turning off "Multi-res Solar System Bodies". This could be an issue with the textures not being loaded from the internet. Also if that does work, turn Multi-res on again and hit Ctrl+F5 and see if that reloads the textures.

erinjohn, 2015-06-15:
We tried flushing cache and reloading Solar System.  Now Europa is still there, but Io is gone.  This is with Multi-Res Solar System bodies *not* checked.  Also tried with Multi-res checked, and Io still not there.
[image]

astrojonathan, 2015-06-15:
You are seeing the Eclipse. Turning Shadows and lighting does not effect
the action of the eclipse. If the moon is eclipsed, it does not draw at
all. Turn time up fast and track on of the moons, and watch them blink like
Chrismas lights.

patudom, 2015-06-15:
I would argue that the moons *should* be drawn when lighting/shadows are turned off, even during an eclipse.  It's like looking at the whole model with the lights turned on in the room.  Io wouldn't just disappear from the model because Jupiter/Sun/Moon all happened to be lined up.

erinjohn, 2015-06-15:
Thanks for those insights, Jonathan.  Glad this isn't an issue with the switch to Azure-- but we'd still hope to see this (eventually) corrected, as Earth's moon is still present when "lighting and shadows" are unchecked.

closed 2015-06-15 by erinjohn