WorldWideTelescope/wwt-windows-client #20: "Look At" - Planet : Mercury shows Moon. Moon is empty. Default install from web Win8.1

Broussard98, 2015-07-02:
Just installed this awesome app for the first time. Seeing a bug on the Explore screen at the bottom where it says "Look At". I have "Planet" selected. Imagery = Mercury shows Moon. Imagery = Moon shows Nada.

egcarter, 2015-07-02:
Are you certain that Mercury is not correct?   It has a lot of resemblance to the Moon.   
As for the Moon, please try this:   click on the Moon from the solar system ribbon at the bottom,  wait as it tracks the Moon, and then stops.   
Then, at the top menus, click "View" -> "Use Layer Manager to Control User Settings"
Go to the list of Layers on the left, 
Then, look at "Sky" -> "3d Solar System" -> "Multi-Res Solar System Bodies".  Uncheck this.  
Does the Moon show up?
Please advise!

astrojonathan, 2015-07-03:
This is actually a problem with Digital Elevation Model data for the moon that has not finished transferring to the new servers. It should be resolved today by noon PST.

 

Jonathan

Broussard98, 2015-07-03:
Seems resolved. Thanks.

closed 2015-07-03 by Broussard98

astrojonathan, 2015-07-03:
Gru and his minions stole it... Actually the Elevation Data for the moon had not yet propagated to all the servers. It has finished now and the Moon should be back.