WorldWideTelescope/wwt-windows-client #21: Style, Interface

Nickesm, 2015-07-03:
Some time ago, I asked about a possible change of style of the WWT's interface, today by looking at the code, I can see why you guys haven't changed this interface.

WinForms.

It's time to switch to a newer platform. More possibilities. How about WPF?

Now that the "Windows 10" will be released, is the best time to launch a WWT application at the store.

(Sorry for my non native English)

egcarter, 2015-07-03:
Nickesm,

Now that the source is open, you are certainly welcome to try!

astrojonathan, 2015-10-26:
Windows App Store apps would have far less capabilities than a Windows Desktop app, especially for Planetariums. Eventually we would like to have a separate app store app, but we decided to invest first in the HTML5 and hope that WebGL capability would give us a universal platform that would require no install. 

If there are developers out there who want to try their had at using the core WWT engine inside a Windows Universal App then you will be happy to know that our SharpDX rendering will support than.

NickeManarin, 2015-11-10:
I was thinking about desktop WPF (not Windows Store).

closed 2019-01-01 by astrodavid