WorldWideTelescope/wwt-windows-client #27: Custom "Galactic mode" button doesn't do anything in Solar System mode.

doctorspaceman, 2015-10-06:
Custom "Galactic mode" button doesn't do anything in Solar System mode.  This is the same behavior as the html5 client.  However, the desktop client doesn't show this button be default (it needs to be created as a custom button).  It should either work or maybe do something to show that it is in active in the current mode (but that might need a new property of the button or something.

mjbecker0, 2015-10-06:
Also can

astrojonathan, 2015-10-25:
Galactic view settings is specific to the Sky view. The sky view can use one of three different view transforms, standard celestial coordinates, local Alt-Az, and galactic coordinates.

Custom buttons act on settings. Settings can always be toggled even when they have no effect in the current mode, when the user switches to the applicable mode, they may want the setting to be in effect.

We don't "gray out" or disable settings that are currently in applicable anywhere in the UI, and they are settings. Actions may be grayed out, but settings never are.

closed 2015-10-25 by astrojonathan