WorldWideTelescope/wwt-windows-client #28: Fade in/out slide elements no longer seems to work

doctorspaceman, 2015-10-07:
Fade in/out slide elements no longer seems to work, which leads to “jumpy” transitions (and the transition options between slides have NEVER worked to my memory)

marksubbarao, 2015-10-07:
The transition options between slides DO work. I just did a quick test of all of them.

Also did a render test to see if the fade-in wasn't working, or if it was just happening very fast. Test elements seem to suddenly appear (happens faster than one frame).

doctorspaceman, 2015-10-07:
We will take the slide transitions off this issue.  It is just about fade-in elements not working.

astrojonathan, 2015-10-25:
When we converted from DirectX9 to DirectX11 the opacity fade in opacity got disconnected from the rendering so it was having no effect. I added opacity to the Sprite2d.Draw() overload and the supporting shaders. The functionality is now restored and it also works with timelines as well, so both the fade-in opacity and timeline opacity are both taken into account.

astrojonathan, 2015-10-25:
Change List is now pushed to the main branch. Issue is resolved.

closed 2015-10-25 by astrojonathan