WorldWideTelescope/wwt-windows-client #35: Solar system should fade out and not appear as a bright gaussian when zoomed out

patudom, 2015-10-08:
This functionality changed as of a couple of releases ago (I'm not sure which), and it has broken some tours that we wrote for educational purposes that allow viewers to travel around the sky to different constellations in Solar System view.

For example, when we wrote this particular tour, this view showed only the Big Dipper.  The solar system did not appear as a large, bright foreground object.

[image]

astrojonathan, 2015-10-08:
When we are inside the solar system the sun is dimmed so we can see other objects without the sun overpowering them, outside the solar system, we brighten the sun to its correct relative magnitude, so it can be seen as it is compared to the surrounding stars.

patudom, 2015-10-15:
It's nice that it now displays the correct relative magnitude, but it's
very frustrating when later builds of WWT with new features like this break
old tours that we spent a lot of time creating and now have to fix.

Is there a way to make this feature an optional setting that we can toggle
on/off?

astrojonathan, 2015-10-25:
The Display of the Sun is now tied to the "SolarSystemPlanets" setting, when you turn off the planets the Sun did not display either, except when you zoom out of the solar system and it faded up to the level of the surrounding stars.

Now the Suns representation will turn off all together when the SolarSystemPlanets setting is turned off.

closed 2015-10-25 by astrojonathan

kfogel, 2015-11-06:
@patudom, did this solve the problem?

patudom, 2015-11-09:
@kfogel, thanks for checking in.  I assume this is a change to the SolarSystemPlanets setting that won't "kick in" until we have a new build.  Until I've seen the new build I won't know for sure if this fixes our tour issues.  It may be that we still have to go through and redo those slides with the Planets button unchecked, but that's easier to do than adjusting all the views to minimize glare from the Sun.  Can you confirm, please, @astrojonathan?  Thanks!

astrojonathan, 2015-11-09:
Just checking in a fix to GitHub is not enough for end-user to get the change. We have to occasionally release a new build and it will include all those fixes. Some data issues, or issues in the WebClient can be fixed without the need for a new setup, but this is a change in the Windows Client.

It might be better to change the slides to adjust for the glare of the sun so the tour will work with code already installed on user machines. Rather than waiting for the new build, and then everyone would need to update it.

The work-around is too zoom a little further out. 

The change to the code was meant to better represent the physical reality of the universe better. We may have other features like the local group, globular clusters, dwarf galaxies, etc. that will change the look of zooming out of the SolarSystem view in the future.

Jonathan