WorldWideTelescope/wwt-windows-client #37: ASCOM telescope control interface not working.

kfogel, 2015-10-22:
Porting this bug report over here from http://forums.worldwidetelescope.org/t/when-will-we-get-a-working-ascom-telescope-interface/20, where it was posted by `tomb18` (not sure if that's the same person as the GitHub user with that username -- Tom, please let us know).

Waving to @astrojonathan here to see if he can answer the question at the end of Tom's report below:

--------------------------------------------------
Every year, I download the latest Worldwide Telescope version and try out the teleacope control interface and every year it still doesn't work! I understand that one of the developers of WWT was actually involved in the ASCOM standard.

So if that's the case why doesn't the telescope interface work correctly?

So this time, the latest release operates like this:

You click connect. It then locks up the user interface for at least 45 seconds. When it comes back, searching for an object and then pressing slew, will slew the telescope fine, but only if track telescope if OFF.

If it is off, the telescope slews fine, but the minute you click on Track Telescope, the image in WWT is about 15 degrees off. However the scope is displaying the image fine, and is tracking fine.

A previous release didnt have a working "Track telescope " at all.

So what should happen: 

1. Clicking on connect, should connect immediately and not stop the user interface. i.e. Ascom should be on another thread than the user interface.

2. Clicking on Slew should slew even in track telescope is enabled. The telescope should slew, the coordinates calculated and everything should be correct.

Now, I would happily fix this myself, since I have written software that interfaces to ASCOM and I also have extensive experience in serial port programming. All I need is to know where the ASCOM code is in the GIT. Any help?

Thanks, Tom

tomb18, 2015-10-23:
Yup, it's me. I'll do some additional trouble shooting.
It would be fabulous to get this working. I'm going to get some logs next week to see if there is an easy fix.

astrojonathan, 2015-10-25:
You should never attach WWT directly to a telescope, always connect thru a Hub, you should not see any delays connecting and the hub will insulate WWT from any communication issues with the telescope.

I have fixed the issue in the Tracking. There was code to convert from RA to LNG that was being called unnecessarily. When the code was originally created it was correct, but the class constructer was later changed to interpret the construction parameters as an RA value if it was of Sky type automatically, so the conversion from RA to LNG before calling the constructor was redundant and caused broke the tracking.

I am checking in the fix.

astrojonathan, 2015-10-25:
Fix to tracking is checked in.
Always use a ASCOM hub when connecting from WorldWide Telescope. Start a hub such as POTH, or configure it from WWT chose, and setup, then connect the telescope to the hub, once connection is validated, connect WWT to the hub. This does two things, it insulates WWT from drivers that hang during communication initialization or loss and it allows other applications to also monitor and control the telescope.

closed 2015-10-25 by astrojonathan

tomb18, 2015-10-25:
Hi,
Thanks for this.  I was unaware that Poth should have been used but that is good to know.
My scope is dismantled at the moment since I am preparing it for being moved outside for the Winter, but should be reassembled in a day or two.  I’ll test that then.
This is great, always wanted this to work right.
Best regards, Tom

From: Jonathan Fay 
Sent: Sunday, October 25, 2015 4:16 PM
To: WorldWideTelescope/wwt-windows-client 
Cc: tomb18 
Subject: Re: [wwt-windows-client] ASCOM telescope control interface not working. (#37)

Fix to tracking is checked in.
Always use a ASCOM hub when connecting from WorldWide Telescope. Start a hub such as POTH, or configure it from WWT chose, and setup, then connect the telescope to the hub, once connection is validated, connect WWT to the hub. This does two things, it insulates WWT from drivers that hang during communication initialization or loss and it allows other applications to also monitor and control the telescope. 

—

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

tomb18, 2015-10-26:
Hello,
All is well. The telescope now tracks properly.
I would like to add an ASCOM rotator to wwt. This would allow setting and synching the camera field of view with  the field of view in wwt. Any suggestions as to an approach? Perhaps as part of the setup dialog for field of view?
Thanks.

Sent from BlueMail

kfogel, 2015-10-26:
@tomb18, you probably found it already, but just in case not: @astrojonathan's diff is commit 4f64750188.  (You had asked where ASCOM code is; at least some of it is in `WWTExplorer3d/TelescopeTab.cs` as per that change.)