WorldWideTelescope/wwt-windows-client #40: Please do not make cursor disappear when tour is running

patudom, 2015-10-31:
To highlight that tours are meant to be interactive, the WWT Ambassadors team has built many tours that encourage people to pan around and zoom in different views, then move on to the next part of the tour.

However, when a tour is running, the cursor will disappear when the mouse is hovering over the top or bottom menu bars, causing problems for slides like the one in the attached tour.  I don't remember this being an issue in older builds.  Can we keep the cursor visible on the whole screen while tours are running, please?

Oops, Github won't let me attach a tour.  @astrojonathan, I will email you an example.

astrojonathan, 2015-11-02:
WWT has always hidden the cursor. Most people don't want to have the Cursor getting in the way of the content while a tour is playing. If you move the mouse it shows up instantly so it can be used if needed.

You have not given me enough information to reproduce unwanted behavior. Can you describe exactly how it behaves and why it is a problem? I have not gotten an e-mail. You can post one in the communities and share it with me.

doctorspaceman, 2015-11-02:
I meant to respond to this. Showing the cursor would be bad for cluster done or kiosks for parts where the tour is not intended to be interactive. 
I don't know if it will work for your purpose but if you have a slide with a start camera but no end camera position it will show the on screen controls which do a good job to invite interaction.  When the slide time is over it continues on and controls are hidden if the slide is narrative with both start and end camera positions.

patudom, 2015-11-05:
I just resent the email to both of you, @astrojonathan and @doctorspaceman.  (I originally sent it on 10/30; it may have gotten lost in long email thread called "weirdness just for you?").

patudom, 2015-11-06:
Let me restate the issue.  I agree that the cursor should not appear while the tour is running.  However, if there is a slide in the tour where the user is explicitly supposed to be able to interact with it (pan around, zoom, etc), the cursor should reappear when the mouse is activated.  This does happen for a lot of the screen, but it vanishes when you move the cursor over the top or bottom menu bars.  For our purposes, it is important for people to be able to make those selections while the tour is running.  If it could be made an option, we would be grateful.