WorldWideTelescope/wwt-windows-client #41: Publish error

tomb18, 2015-11-02:
Hello, I am having an error if I "Publish" a downloaded zip file of WWT.
I am using VS 2013 (same happens in VS2015).  In both cases the log file shows - System.IO.IOException
		- The file 'C:\Users\Tom\AppData\Local\Temp\Deployment\D0OWMHBD.LEJ\DRO6E562.Q98\DSMRender.dll' already exists
Does anyone have any ideas?
Thanks

astrojonathan, 2015-11-02:
What do you mean by publish? Do you mean the Visual Studio Publish Menu?

The WWT client is a Windows app. You Build it and include the setup build image, and that .MSI file is your installer.

I am not sure what you are trying to accomplish?

tomb18, 2015-11-03:
HI, I am trying to produce an MSI or executable so that I can run my modifications on another computer of mine without having the VS 2013 installed.  So I guess I am not sure how to do so,  I tried to produce a Click once install by using the visual studio Publish menu command.
I can build the WWT client in debug only.  Release does not work.  
In my experience (limited...) the visual studio Publish command provides a Click once install file.  That is what is not working.  I get the error above.
Now if the build also provides an MSI file, that I was unaware of.  Where would this be found?
On another note, I have added an ASCOM camera rotator to the telescope menu which allows rotation of a camera if you have an ASCOM driver for it. The screen field of view indicator updates accordingly. I guess I can make this available to the GIT when this  becomes open.
Thanks.

tomb18, 2015-11-03:
Hi,
I just checked....no MSI file is created when a build is performed.

astrojonathan, 2015-11-03:
You need to have the Visual Studio Installers add-in installed to build the Setup project, this is in the readme on the main GIT root. One you are ready, select Release for the build type and right click on the setup project and click Build. The setup will be in the Setup1\release directory as an MSI file.
The setup is well tested and known working, you just need to figure out how to find in and have the right prerequisites installed.

astrojonathan, 2015-11-03:
Also you can do a pull request on changes you make, this is open source, but its not a free-for-all where anyone can check-in code to the main  repository, you make a pull request with your changes in GIT and then we review and see what we can take and what we can't.

closed 2015-11-03 by astrojonathan

kfogel, 2015-11-06:
Well, true, but then again no open source project is that kind of free-for-all -- taking pull requests is the norm everywhere.  @tomb18, it sounds like if there's a PR to be made here, it would be an improvement to the build/install documentation?  Was there something missing in the docs?  It's fine to just point it out here too, rather than make the full pull request -- do whatever's easiest for you.