WorldWideTelescope/wwt-windows-client #42: Merge tour after slide... bug?

patudom, 2015-11-05:
On 2 different computers earlier today, our team experienced a bug with "Merge tour after slide."  (in build 5.2.9.1).  Instead of merging the selected tour into the working tour, WWT merged the working tour onto itself (and created a duplicate set of slides.)  When we clicked control-z to "undo," ALL the slide thumbnails disappeared from the tab, but the tour still ran.

I'm trying now on a 3rd computer running 5.2.9.1, but the merge seems to be working, so I'm not sure what's causing the problem, or how easy it will be for someone else to reproduce/troubleshoot.

patudom, 2015-11-05:
I just reproduced the bug on the 3rd computer.  This is happening specifically with 2 sets of slides that started out as being the same tour.  (One of our interns added new slides to the end of our working tour draft, which we wanted to merge into a different version of the same tour in another file.  She deleted all the slides the 2 tours had in common from the file to be merged.  But when we tried to merge her new set of slides into the working draft, it just duplicated the original set of slides.  I suspect there is some issue where there  is identical meta data in the 2 tour files (because they originated from different drafts of the same tour), and the merge code is getting confused about what to put where?

@astrojonathan, let me know if it would help to send you the 2 files in question.

(When I tried to duplicate the bug on a different computer, I didn't have exactly the same 2 files that caused the problem in our group meeting today, so I tried merging 2 completely different tour files, which worked fine).

astrojonathan, 2015-11-09:
The issue is the tour thinks it is merging with itself.  I will work on a fix.

closed 2019-04-15 by astrojonathan