WorldWideTelescope/wwt-windows-client #43: Time for ISS Spice Kernel

doctorspaceman, 2015-11-12:
When we did the Impacts tour we had an issue the location of the ISS was changing between times of playing tours because the revised positions were updated in real time compared to what was produced for.  This seems to have come back or at least something changed between when the tour was created for Summer 2014 and now (fall 2015).

astrojonathan, 2015-11-12:
Delete the ISS reference frame before you load it, or load it via double
click from the browser.

When loading tours as a player only, the reference frame is used from the
tour, if you open of edit, the current ISS reference frame is used, so you
have to delete it before loading to make sure you use the one from the tour.

Jonathan

astrojonathan, 2015-11-12:
BTW. We use TLE orbital elements, not Spice Kernels.

PlanetariumSU, 2015-11-20:
Perhaps it would be good to add a way to import satellite data relative to a body without using a TLE, since Spice kernels take orbital maneuvers into account. In my own experience, I'd be thrilled to add Rosetta orbiting 67P, a host of satellites around Mars, Dawn around Ceres and Vesta, etc

astrojonathan, 2015-12-03:
I should have been more specific. We don't user Spice kernels, we support properly described Keplarian orbits including TLE's, but 6 "orthogonal" elements will do.

Also we have support for trajectories in reference frames so you can user pre-computed trajectories generated by an outside SPICE Kernel tool and import them into WWT.

closed 2016-06-09 by doctorspaceman