WorldWideTelescope/wwt-windows-client #45: 3D Solar System Orbits Clipping for trans-Neptunian objects

AstroRob, 2016-01-13:
I’m working on a project where I’m trying to show the distant bodies in the Solar System in context (e.g. Sedna, and some of the other trans-Neptunian objects). I added the minor bodies orbits as layers, looking them up from the MPC. 

They display great as long as I stay near the solar system, but when I push further out to show the entire orbits within the field of view there seems to be some flickering clipping that shows up on the far side of the sun. 

Oddly the distant clipping limit seems to get geometrically closer to the sun as I pull the camera further from the sun. In other words, if I’m in the solar system there seems to be no clipping out to hundreds of AU, but if I move hundreds of AU away, then I seem to lose orbits that extend behind the solar system.

These are some examples that show the issue:

Sedna
2007 TG422
2010 GB174

Note that the issue is definitely rear-plane clipping. If you’re more or less in the plane of the orbits as they stretch out behind the sun (where they display fine), then zoom out to a more distant perspective, that’s when they start clipping (like in the screen capture I sent). The further out you zoom, the more the larger orbits get clipped.

It never gets tight enough to affect Pluto/Neptune

astrojonathan, 2016-01-13:
I looked into this and it seems that when you are zooming out it actually does some math that incorrectly handles the transition case from inside to outside the solar system mode and will clip orbits outside.
I will check in a fix tonight.

astrojonathan, 2016-01-13:
Fixed

closed 2016-01-13 by astrojonathan