WorldWideTelescope/wwt-windows-client #50: Error starting WWT (Windows 10 V1511)

Basti1512, 2016-01-26:
After a fresh installation (first time) I'm not able to start WWT. Everytime I've got the following error message
[image]
I'm running Windows 10 V1511 with 8GB RAM and 1920x1080 resolution. A re-installation doesn't help.

cecilia-donnelly, 2016-01-26:
Hey @Basti1512, thanks for the bug report.

@doctorspaceman, does the desktop version of WWT work on Windows 10?  Your answer [here](http://forums.worldwidetelescope.org/t/how-to-download-wwt-using-windows-10/24) suggests that it does not.  On the [download page](http://www.worldwidetelescope.org/Download/), Windows 7 or 8 is specified.  The text says that older Windows versions don't work, but doesn't say anything about newer versions.  If WWT fails with Windows 10, that should be stated explicitly -- I'll file a bug in [the website repo](https://github.com/WorldWideTelescope/wwt-website) to update the text on the download page.

astrojonathan, 2016-01-26:
WWT runs *best* on Windows 10. There are a couple of issues that can cause this problem, aborting the very first startup, or a corrupt system configuration.

First step is not click OK. Click the other option.

Second if that does not work then you may need to complete remove two directories.

Substitute your user name for the {user} part of the path.

C:\Users\{user}\AppData\Local\Microsoft\WorldWideTelescope

and any directory that has WWTExplorer.exe in it

C:\Users\{user}\AppData\Local\Microsoft_Research*

After you do that try running again.

If it still does not run it may be due to missing .NET framework, no DirectX support or other missing pre-requisites.

astrojonathan, 2016-01-26:
I removed the first reply. It had jumped to some incorrect implications. WWT is fully supported on Windows 10.

cecilia-donnelly, 2016-01-26:
:+1: thanks for the clarification @astrojonathan

Basti1512, 2016-01-26:
The error message still occure by starting WWT the first time (however), but it helped not to click on OK. (I tried CANCEL already, but not at the first step. After clicking OK once, CANCEL doesn't help anymore.)

Thanks a lot for your help.

astrojonathan, 2016-01-26:
Please delete the data directories as stated in my first response.

Basti1512, 2016-01-26:
I already did and the problem was solved with my previous comment. Sorry for the misunderstanding.

astrojonathan, 2016-01-27:
Not a Code Issue. Closing

closed 2016-01-27 by astrojonathan