WorldWideTelescope/wwt-windows-client #56: Please support Consumer Version of the Oculus Rift

LeeHolmes, 2016-04-11:
As https://github.com/WorldWideTelescope/wwt-windows-client/issues/44, now that the Consumer Version is out, I'd love to see support for it in WWT!

astrojonathan, 2016-07-24:
I received a Consumer version this week, and Today I ported WWT to it. I just got it running and it works really well. I need to do a lot of testing on it before I check it in. It required a rewrite of the Oculus layer. Every time they release a new SDK to support a new device, it breaks the old one!  This is the third Oculus Implementation in WWT. DK1, DK2 and now CV1.

The head tracking is solid even when WWT can't keep the framerate at 95hz. It does Asynchronous warping of the last good frame, for each and every frame. This means you never, ever see a freeze in the rendering. It is sooo much better than the DK2, which was worse than the DK1 in terms of performance.

The new SDK may also mean that the older units work better as well. I need to check out the DK1, HD and DK2 out with the new SDK and see if they are supported.

LeeHolmes, 2016-07-24:
Awesome, can't wait to experience it!

VeryKross, 2016-09-15:
@astrojonathan is your branch with the consumer Oculus changes available? I have a consumer edition as well and would love to try it out - right now WWT doesn't seem to want to connect to my Oculus (always complains it's "in use"). Hopefully this is the sort of thing your fix addresses!

astrojonathan, 2016-09-15:
I have to fix overlays in tours (the new stereo rendering screwed up their alignment) but then I can check them in. I have bee pre-occupied with the new Web Client Tour Editing and WebGL the last few weeks.
Will try to get this in tonight.

VeryKross, 2016-09-15:
That would be awesome! Don't kill yourself though - if it's available by the weekend that will give me plenty of time to pull it down and check it out. :)

astrodavid, 2016-09-27:
Any updates on this? We are planning a VR experience day on Oct 29th and would love to use the highest res headset possible

astrojonathan, 2016-09-28:
Oculus updated the runtime and it broke the implementation that I was working with.  I have managed to fix it but tour overlays are still not registered properly in the stereo view. I can put a temporary build with the fix up for you to try, but its not ready to release until I fix the overlays.

astrojonathan, 2016-09-28:
Got the overlay issue and the version mismatch issue fixed. Here is a URL for a test build of the Oculus CV-1 version. Please try it out and let me know how it works.

https://1drv.ms/u/s!AgURmnvmzM_MpvFFn_3n5_YXAUgvXw

astrodavid, 2016-09-29:
Awesome, thanks! I'll check it out and get back to you next week!

astrodavid, 2016-10-03:
After a quick few hour test, things look phenomenal compared to the DK1! We're running into an issue with one of our VR tours, however. When we start rendering terrain on the earth the frame rate slows to a crawl and playback is very choppy. I've attached the .wwt file via onedrive for the tour for reference. Is there an arbitrary buffer size built into WWT that can be turned off? Our target framerate is unlimited as is our tile throttling. Anti-aliasing is also maxed. Our computer has an Intel Xeon quad-core 3.6 GHz CPU, a GTX 1080, and 64GB of RAM; nearly none of those resources are being used. Thoughts @astrojonathan ? Here is the onedrive link: [https://1drv.ms/u/s!Ak8mOJrdjGOxtTJeGU6g6ob7lhSV]

AstroRob, 2017-06-01:
Bumping this thread, we just got a new Alienware/Oculus rig and I have been unable to get WWT to function with it. I've downloaded the build Jonathan posted on 28 Sept and tried to follow the sequence on the WWT homepage for Oculus.

When I go to activate Stereo/Oculus Rift option the program crashes with the usual error crash dialog. If I go to the Oculus mono mode option then nothing happens (the Oculus just shows the usual storefront app). If I then select the stereo mode, it crashes.

It also crashes if I select the Oculus on startup option and restart the app.

I'm not sure if there is a workaround, or if there might be new issues with recent Oculus updates. This is brand new equipment just out of the box.

astrojonathan, 2017-06-01:
Rob, I send you a link to a new build. There have been delays in getting the official builds out as the AAS program management team was shifting leadership. Phil officially starts June 1st and we will have officially sanctioned builds of the changes in GitHub on the AAS/WWT website for download soon.

In the meanwhile leave me comments on the build I sent you.

Jonathan

astrojonathan, 2017-06-01:
When the new official build of WWT comes out only the Oculus CV1 will be officially supported. You might be able to get the DK2 to work with some hacking, but the DK1 and HD are confirmed to be NOT supported. We will also support the hand controllers in a future release, and those won't work on the DK2 at all, so the everything prior to the CV1 will no longer be supported on new WWT builds.

cynicsymmetry, 2017-09-08:
Hi @astrojonathan! I work for the University of Arizona, and one of our department heads recently purchased an Alienware computer and a CV1 specifically to use World Wide Telescope for a research grant. I have gotten it to work with 5.5.20 posted earlier, but I wanted to see if there was a newer test version available, or what the timeline for the official release looked like. Also, what are the chances of that version of working with Oculus Touch?

Thanks!
Anthony

astrojonathan, 2018-01-26:
We ported WWT to Windows Mixed Reality Headsets with the motion controllers. On that same branch I am working on full Oculus Parity for controls as well. So the same functions available on the Windows Mixed Reality controls will eventually be supported in Oculus in the Windows Client 6.0

reidma, 2018-06-07:
@astrojonathan can you comment on the current status of officially supported VR headsets for use with WWT? I see that WWT v6.0 is not out yet and the WWT site only mentions official support for Oculus DK1. Is there official support for the Windows Mixed Reality headsets, Oculus, or any other headsets?

astrojonathan, 2019-04-15:
Oculus CV1 & Touch controllers are supported in 6.X release.

closed 2019-04-15 by astrojonathan