WorldWideTelescope/wwt-windows-client #59: While loading a FITS image only the main header is checked and not the Primary HDU header

doctorspaceman, 2016-05-10:
NOTE: This issue was contributed to the "WorldWideTelescope/wwt-web-client" repository (Issue 92) but should be here (Windows Desktop client).

Some FITS files (like the ones generated by the Herschel mission, see attached file), do not have NAXIS, CROTAi, CDELTi, etc. in the main header, but in the Primary HDU header.
Since WWT only checks the main header of the FITS file (until the END keyword), these FITS files cannot be loaded by the program and they fail with the error message: "The image file did not contain valid AVM metadata to position it in the sky".
The reason is that:
NumAxis becomes 0 because NAXIS is in the primary HDU header (not parsed).
WCS is only computed if NumAxis > 1.
Valid AVM means valid WCS.

[test.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/257485/test.zip)

harcalion, 2016-10-13:
I have an initial patch for this. Will test with additional FITS files (HST, for example) and then submit a pull request.

harcalion, 2016-10-26:
See pull request #83