WorldWideTelescope/wwt-windows-client #64: Orbit not showing for MPC reference frame

doctorspaceman, 2016-05-13:
I created a reference frame for an MPC object,  By default it shows the orbit line and the dot at the head.  I unchecked the orbit line box in the reference frame properties.  The orbit line went away.  Then I checked it again and it didn't return.  

I created a new MPC reference frame object for a totally different body.  It also is in a state where I can't get the orbit line to show even though the box is checked.  I'm including a layer file for this.

[Vesta_no-orbit.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/264226/Vesta_no-orbit.zip)

astrojonathan, 2016-05-14:
I have no trouble viewing it.

doctorspaceman, 2016-05-14:
My mistake.  I had "Moon & Satellite Orbits" turned off.  User error.

closed 2016-05-14 by doctorspaceman