WorldWideTelescope/wwt-windows-client #74: Tours should set whether Volumetric Milky Way is shown

patudom, 2016-07-06:
In the 3D Solar System view, tours should be able to set whether the Volumetric Milky Way is shown.   For example, if you're flying around constellations in the Hipparcos view, the fuzz from the volumetric Milky Way can obscure the view of the stars.

When you run a tour that has these Hipparcos views, whether the volumetric Milky Way appears seems to be based on what has been set locally by the user.  (i.e. if the user happened to uncheck volumetric Milky Way before running the tour, it does not appear.)  The tour author should be able to override this.

astrojonathan, 2016-07-24:
The volumetric milky way will not show, if the Milky way setting is turned off.
The reason it is not on by default coupled with the milky way setting, is that some systems just can;'t handle the rendering. If I turn the rendering off I don't want a tour to turn in back on. Ever. Because I won't have a good experience.

Just turn off the milky way if you don't want the side effects.

patudom, 2020-04-06:
I agree it's not a good idea for the tour author to be able to turn the volumetric Milky Way on, for all the reasons you list, @astrojonathan, but can we still allow the tour author to turn it off, just for the duration of the tour playback?  

We have a tour where the user won't be able to see what we are trying to show if the user happens to have the volumetric Milky Way on, and many of our users won't recognize that they should turn it off (or know how) before they run that particular tour.

philrosenfield, 2020-05-31:
Has this request been rejected? Is there a workaround in communicating to the user that they should turn something on? Can this be closed?

patudom, 2020-06-01:
If possible, I would like for the tour author to be able to force the volumetric Milky Way setting to off for tour playback, but then put it back to whatever the user had it set at when the tour is finished playing. (I agree that the tour author should not be able to force the volumetric Milky Way to be on).