WorldWideTelescope/wwt-windows-client #79: Alt-Az grid shown in tour breaks from v.5.2.9.1 to v.5.5.03

patudom, 2016-08-11:
We are working on a Seasons tour for one of our NSF projects and need for tours like this to work reliably for all participating teachers.  We're trying to show the path of the Sun from sunrise to sunset as viewed from Fenway Park in Boston, and we want the alt-az grid as a reference frame. 

When viewed in v.5.2.9.1, all is fine.  

[image]

 When viewed in v.5.5.03, the alt-az grid is warped and incorrect.  
[image]

This can be fixed if you put the tour in edit mode and then click "Show start camera position."  Once you do that, it sets correctly and the tour runs as it should.  However, we cannot rely on every teacher having to do this on every computer in a classroom.  Please help?

astrojonathan, 2017-06-13:
This is probably because the location of the viewing site is different on the two computers.

patudom, 2017-06-20:
I will check this, but we don't usually change the viewing site on any of our computers.  They pretty much always stay on the default, and we build the tour on the same computers that we use in the schools.  Also, do you have thoughts on why it used to work with no issues in v.5.2.9.1?  Thanks, Jonathan!

patudom, 2018-05-24:
I get what you're saying about the viewing location, @astrojonathan .  In order for this to work consistently, the tour needs to set the viewing location when it's being run (because we need the alt-az grid over Boston, not some other random place).  Is it possible that v.5.2 took the viewing location input from the tour, but it was changed in 5.5?  Can this be changed back, please?

astrojonathan, 2019-01-05:
Fixed. When playing a tource the space-time controller uses the location from the active tour settings.

closed 2019-01-05 by astrojonathan