WorldWideTelescope/wwt-windows-client #82: 2-Channel Dome Setting.

runfar, 2016-09-29:
We will attempt setting two-channel planetarium dome by using two projectors. Is it possible?
Please guide us how to set it.
Thanks.

astrojonathan, 2016-09-29:
It is possible, but it takes two cropped fisheye projectors with overlap, and two computed cropped fisheye warps to make it work. There is some code to designed to get a best fit warp and translation of warped fisheyes for E&S systems, as far as I know only one planetarium is using it.

What type of projectors are these? What type of lenses? Are they setup for another system already? Do you have warp maps?

I can try to walk you thru it.

Jonathan

runfar, 2016-09-30:
What type of projectors are these?
-This is Sony 4K vw520

What type of lenses? 
-It is 3.5X  conversion lens.  
 (https://navitar.com/products/projection/35x-fisheye-conversion-lens/)

Are they setup for another system already? 
-No. This is first time.
  
Do you have warp maps?
-No. We don't have it.

astrojonathan, 2016-10-31:
I am not sure how well the fisheye calibration built in to WWT will work for that. I am working on another project to do photographic calibration that will output a format WWT will read, and updates allow WWT to output calibration patterns, but it won't be ready for a while, and I am first getting non-fisheye calibration working. I can update you about it when I know more. We could take a shot with the existing code in WWT, but there is not guide on it so I would have to walk you thru it.
Contact me if you want to try it.