WorldWideTelescope/wwt-windows-client #84: Stereo mode crash application

doctorspaceman, 2016-10-15:
When activating  all stereo modes (except Oculus) WWT immediately crashes:
- Analglyph (Red-Cyan)
- Analglyph (Yellow-Blue)
- Side by Side Projection
- Side-by-Side Cross-Eyed
- Alternating Lines Odd
- Alternating Lines Even

- Oculus Rift works fine.

astrojonathan, 2016-10-31:
This was a side effect of the DK2 changes when the depth buffer size was not being adjusted properly.
Fixed in 10/12/2016 in the codebased.

closed 2016-10-31 by astrojonathan

rinax67, 2018-01-15:
Hello. 
I'm Rinaldo from Italy.
I downloaded last version  of WWT bit still have same  problem.
How can I do?
When activating all stereo modes (except Oculus) WWT immediately crashes
Please have you any  installation  file?
Thanks a lot
Rinaldo