WorldWideTelescope/wwt-windows-client #86: Tracking student responses in WWT tours that use linked slides as "quizzes"

patudom, 2016-11-14:
2-part plan to log student responses by tracking "clicks" in WWT tours that use linked slides as quizzes.

Part 1:
- add an optional attribute to the overlay to indicate which answer a click there represents

- add a slide number attribute to tour stop

- output the time, slide number, overlay clicked and correctness to a file which instructor can download off the student's desktops.  (If this solution works well, ultimately we will explore a solution that uploads the data to a server, so we don't have to grab all the output files off the computers.)

Part 2:
- add the ability to indicate correct answers via the tour editor

portermahoney, 2016-11-29:
I created a separate ticket for Part2.  Work on Part1 is done pending the chance to get feedback from @patudom

astrojonathan, 2016-12-23:
I reviewed the pull request and except for using the MAC address, it looks OK. I will build and run it and try it out as well. Do you have usage instructions?

portermahoney, 2017-01-19:
resolved with commit 15d10467699b05aee8d49e77306b23937a61291a

closed 2017-01-19 by portermahoney