WorldWideTelescope/wwt-windows-client #91: Orbital Ref Frame Units

astrodavid, 2017-04-07:
When I create a new reference frame (orbital) and manually fill in the TLE to specify semi-major axis, if the units used are not meters, the orbital line will show. However, once a toasted texture or 3d model is added to the new ref frame, the visualization breaks and a black void is encountered (this sounds strange and it is). I tried this while trying to recreate the Trappist 1 exo-system.

pkgw, 2021-12-03:
Bumping this issue as per recent conversation with @astrodavid10, CC @Carifio24 . Not 100% clear that this issue is talking about the same thing, but since it's been stalled for so long, let's hijack it for the issue that **reference frame setup doesn't honor any units besides meters**.

High priority since, as David mentioned, this can be super confusing and if you don't know that this is a problem it's very hard to figure out what's going on.

Carifio24, 2021-12-09:
@astrodavid @pkgw Do either of you have a specific example of this making something go wrong? I've started looking into this, and having a good example would be a big help.

astrodavid10, 2021-12-09:
@Carifio24 @pkgw In my initial attempt to make you a dataset that is displays the broken functionality I've stumbled upon something related and confusing. It seems that if I create a time, RA, Dec, dist text file and try to add to the layer manager by right clicking the Sun, then selecting add and then finding my file. I am able to see a line drawn that correctly works using AU for units. However, the line is clipped depending on my angle of viewing, especially close viewing.

The first picture shows the full trajectory:
[image]
.

The second shows it clipped at the same zoom level, but rotated around:
[image]

The third, shows it zoomed slightly and clipped on both sides:
[image]

Finally here is the layer file:
[nh_text_test_au.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/7688984/nh_text_test_au.zip)

I'll try as an .xyz trajectory file later tonight.

astrodavid10, 2021-12-10:
.xyz trajectory seems to work with units=km, a sample is attached. No clipping occurs either.
[nh_xyz_km.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/7691189/nh_xyz_km.zip)

I'll try a few more options tomorrow.
.

pkgw, 2021-12-10:
@astrodavid10 Hm, interesting. Maybe related to the "far side visible" setting? But it would probably be good to stay focused on the units issue here, if you're able to give a quick recipe for reproducing that part. Task-management-wise, discussion about the Z clipping should go into a different issue.

Carifio24, 2021-12-13:
@astrodavid10 I've been working on this and just want to check in that I'm on the right path here. In both images, the white orbit is Earth's.

Here's the trajectory that you sent, using meters as the reference frame units:
[image]

and using kilometers:
[image]

pkgw, 2021-12-14:
@Carifio24 Based on David's filename I think this is a trajectory for New Horizons mission that flew by Pluto, so the one with km units looks correct — probes are launched from Earth('s orbit), after all. The trajectory should intersect Pluto's orbit ellipse if that's all correct.

astrodavid10, 2021-12-14:
@Carifio24 @pkgw The km units are correct (and New Horizons flew by Pluto on July 15, 2015 so setting the time to that date should allow us to se an intersection with Pluto at that date if all is correct).

I have tested a few units with km first and meters, AU, and PC don't work. Meters are too big and the trajectory line ends up outside the SS, a factor of 1000 too large. Both AU and PC are too small and are inside the Sun. KM are the only correct units. I wanted to see if it was dependent on the first specified file so I tried again loading other units first and the results are the same. I also tried without specifying units or selecting the wrong units and it seems that the unit selector does not function as expected.

[nh_xyz_tests.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/7708707/nh_xyz_tests.zip)

These xyz file assets can be loaded by right clicking the Sun -> new ref frame -> offset type = trajectory -> show orbit path and point at distance -> import path (select units does not seem to work)

Carifio24, 2021-12-14:
@pkgw Ah, I had not made the connection that NH = New Horizons. And yeah, with the units set to km, that trajectory is intersecting Pluto's orbit.

@astrodavid10 From looking at the code, everything that you're saying tracks with what I'm seeing. Currently, the units are fixed as meters, but there's a hard-coded factor of 1000 inside trajectory positions - this is why only kilometers work. I should have a PR up soon that addresses these issues that you can test out.

Also, thanks for the tests. In my working version, they're overlapping when I set the units appropriately, so that's a good sign!

astrodavid10, 2021-12-14:
@Carifio24 That is great to hear, looking forward to it!