WorldWideTelescope/wwt-windows-client #100: Add ability to export a cubemap in equatorial projection

astrodavid, 2018-01-12:
So that we can render out VR tours to videos for youtube! Preferably with the option to specify resolution so that we can push 8k+ videos. Will need to be able to render tours when not in VR mode.

astrodavid, 2018-07-16:
Bumping this again based on the amount of interest expressed by different individuals at IPS 2018

pkgw, 2021-12-03:
And bumping again. Flagged by @astrodavid10 as something to look at for the upcoming release. CC @Carifio24 although this is definitely a wishlist item.

There are probably a lot of ways that one could approach this, but IMO the way to start would be to laser-focus on creating output that can be uploaded into YouTube 360 videos.