WorldWideTelescope/wwt-windows-client #106: Ability to create a "New" Constellation

astrodavid, 2018-03-20:
Currently there is no way to add an additional constellation, more than the standard 88. The only way to add asterisms or other cultures' constellations is to overwrite an existing constellation (and these cannot be renamed).

philrosenfield, 2018-03-20:
The Chinese VO, for example, has used the excel plug-in to make constellation layers (poly lines) as a work around.

However, it would be beneficial to have features that allow users to add constellation figures, text, and images and not need a work around (or perhaps still enable focus only).

Let’s use comments in this issue to brainstorm the details of the enhancement and tag in others who may be interested.

astrojonathan, 2018-03-24:
The 88(9) constellation defined by the IAU are specific area of sky. If you want to make a asterisms layer you could create a layer group, and a number of WKT layers for Each asterism by name. You could also have text labels. In fact this was a exercise in a Adler Planetarium Sumer camp a few years ago.

We are not going to "Add constellations" to the canonical IAU list.

philrosenfield, 2018-03-24:
I have heard multiple requests to have access to different constellation sets from different cultures. 

I didn't mean to suggest to add constellations to the IAU list. I was asking to add a set of constellations to use instead of, or in comparison to, the IAU list. That way, they would have the same functionality (except boarders) as the current constellations.