WorldWideTelescope/wwt-windows-client #108: Add Ability to Control Astrophotography Overlays in the Timeline

astrodavid, 2018-03-21:
Currently there is no way to crossfade different astrophotography images in the timeline and it would be excellent if added.

astrojonathan, 2018-03-21:
You can have several Image Layers, and then fade between them. The single forground layer is for quick and dirty work, but image layers are the workhorse for TimeLine Tours.

closed 2018-03-21 by astrojonathan

astrodavid, 2018-03-21:
If I add an image as a layer it becomes a child of "Sky" but I don't see a way to add those to the timeline. Any advice?

astrodavid, 2018-03-21:
I am able to set a beginning camera start with no opacity and then an ending camera setting with full opacity and this completes the fade.

astrodavid, 2018-03-21:
Important to note that these fade transitions must be their own slides and cannot have a timeline created for them. This looks nice when all is said and done.

astrojonathan, 2018-03-21:
Well there is a bug right now where ImageSet layers are not having the Add to Timeline/Add Keyframe menu item show up. 

I just checked in a fix. It only effects authoring.