WorldWideTelescope/wwt-windows-client #109: Camelopardalis Constellation Lines Missing

astrodavid, 2018-03-24:
While I'm thinking about it, Camelopardalis is missing its constellation lines (boundary and picture exist), which are easy enough to add to WWT but should be a permanent fix.

astrojonathan, 2018-03-24:
There are few naked eye stars and not a lot of canonical lines connecting them in Camelopardalis when we created the constellation. Aside from art work I could not find any lines that two sources agreed on at the time so we had no figure for it. The constellations are defined by the outlines, and not their figures, but many constellations have asterisms that are popular enough to be somewhat agreed on.

astrodavid, 2018-03-27:
I used the IAU defined lines as shown here: https://www.iau.org/static/public/constellations/pdf/CAM.pdf