WorldWideTelescope/wwt-windows-client #111: Layer Manager triggers Auto Hide Tabs bug

patudom, 2018-05-24:
If you hide the left hand layer manager box by clicking the "x" in the upper right hand corner,  "auto hide tabs" stops working, even when a tour is running.  The only way I've found to make the top menu bar hide again is to restart WWT.

(If you click on the "foot" in the lower left corner to make the layer manager menu go away, the autohide feature continues to behave properly.)

patudom, 2021-12-06:
@pkgw This feels like one that ought to have a straightforward fix.

pkgw, 2021-12-06:
Thanks for bumping! CC'ing @Carifio24 here again.

Carifio24, 2021-12-09:
This sounded like it'd be very simple to fix...which turned out to be the case! See #182.

closed 2021-12-09 by pkgw

pkgw, 2021-12-09:
@patudom If you go to [this link](https://dev.azure.com/aasworldwidetelescope/WWT/_build/results?buildId=2294&view=artifacts&pathAsName=false&type=publishedArtifacts), are you able to click through and download files? If so I can point you to how to download an installer and test out the new version. If not we can get you the file another way, if you're interested to test out the fix.

PR #182 (2021-12-09): Fix issue with auto-hiding tabs
This PR fixes #111, which is that the Auto-Hide Tabs functionality does not if the LayerManager is closed via its close button.

The cause of the problem is the static variable `LayerManager.InsideLayerManagerRect`, which keeps track of whether or not the cursor position is within the layer manager. When the layer manager is closed, this variable no longer gets updated, so closing the layer manager using its close button (when `InsideLayerManagerRect` is `true`) means that the app continues to think that the mouse is over the layer manager. The fix in this PR is [... cut]