WorldWideTelescope/wwt-windows-client #118: Some Problems about Make Tour

qu123xxx, 2018-08-27:
I'm using WWT to make a complex tour,but I have several problems:

1. The most important is the "Reference Frame" ‘s “Opacity” or "Color Alpha" can not be controlled in "Timeline Editor". 
For example ,if I have Reference A and Reference B under the Sun，then the "3D solar system" interface will  show them in all the tour. But I just want to show A or B in one "Silde" and show the other one in other "Slide" .

2.Some colors can't show in the slide. Such as purple、blue、pink
This bug was found when I create a silde to show star chart and target it as pink. It became red when the slide was created . Color bule will become black and you will see nothing in the windows.

astrojonathan, 2018-08-27:
Can you give me a link to the tour with this issue so I can look into it?

qu123xxx, 2018-08-27:
Thanks for  your reply.
This is the example and describtion of operation of the tour： 
https://1drv.ms/u/s!AnBkLwQlyfVwlw00bAX5aSvX6Rk1

qu123xxx, 2018-08-27:
Thanks for your reply.
This is the link of the example : https://1drv.ms/u/s!AnBkLwQlyfVwlw6Kn-TP0wGYQXoN
There are descriptions and the steps of how to repeat the problem in it.

philrosenfield, 2020-05-31:
Closing. Example links are broken, feel free to reopen or discuss issues [on our forum](https://wwt-forum.org/)

closed 2020-05-31 by philrosenfield