WorldWideTelescope/wwt-windows-client #123: Pluto system Inaccuracies

astrodavid, 2018-10-31:
Missing Pluto (and Charon) new horizons imagery.
Pluto rotates backwards on its axis.
Charon and Pluto are not tidally locked to each other - I'm not sure of the reference frame, so I have yet to be able to get this fixed on my PC.
Missing Kerberos and Styx orbits. 

The end result will look great: https://www.youtube.com/watch?v=VAGBPbToZYE

astrodavid10, 2020-06-02:
Should note that when you overlay the lat lon grid on Pluto, the imagery aligns correctly. Pluto's heart should always face away from Charon. Charon's North pole incorrectly wobbles as it orbits Pluto.

forceflow1049, 2020-07-16:
I want to bump this report. I'm working on a project that for now has to use the Windows client and it's frustrating that the Pluto imagery isn't up to date when it is on the web client.

pkgw, 2020-08-05:
@forceflow1049 I hear you. Work on the new Windows release has slowed with the pandemic, and the full data update won't come until we finish the round of beta releases, but we should make sure to fix as much of this as we can.