WorldWideTelescope/wwt-windows-client #128: Spherical Mirror Warp Regression 6.0.4.1

astrodavid, 2019-01-19:
In version 6.0.4.1, the spherical mirror warps, 16x9 or 4x3, are zoomed in on the usual view which distorts the warp.

pkgw, 2019-01-21:
Thanks. This is basically just my ignorance, but can you explain to me what you mean when you write that the views are "zoomed in on the usual view"? Thanks.

astrodavid, 2019-01-22:
Of the three issues I just opened, this is by far the most pressing as I will not be able to use the release in a planetarium mode if this isn't fixed :/ I'll post tomorrow a picture of what it should look like (when I get back to my other computer), but basically we are only viewing perhaps 80% of the usual view, it is cropped on the sides and bottom. All of the Az lines in the image should bend down to the ground.

astrodavid, 2019-01-22:
No worries! Here is the proper warp from a previous build (5.5.20.1)

[image]

astrodavid, 2019-01-22:
Fixed.

closed 2019-01-22 by astrodavid

pkgw, 2019-01-22:
Code change, or did the problem not end up being on the WWT side?

astrodavid, 2019-01-22:
Slight code change: https://github.com/WorldWideTelescope/wwt-windows-client/commit/11a53e1c4f9e4b0f5633cf9a3f94f738ca681df6

pkgw, 2019-01-22:
Great, thanks — it's helpful to cross-link the issues and the code (exactly like you just did!) for posterity.

astrojonathan, 2019-01-22:
The issue number is usually included in the PR description, but I missed that.,