WorldWideTelescope/wwt-windows-client #129: Local Group Data disappears close to Milky Way

astrodavid, 2019-01-19:
Similar to the clipping that occurred for Planet 9 orbits, data points for the local group disappear near the Milky Way. WTML file attached,
[Local Group.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/2775139/Local.Group.zip)

astrodavid, 2019-01-19:
@patudom

pkgw, 2019-01-21:
Thanks David. Presumably you'd like to have this fixed in the upcoming Windows release? How urgent is this to you?

astrodavid, 2019-01-22:
If it is easy, great! If not, it isn't very pressing.

astrojonathan, 2019-01-22:
This is not a code bug. You need to turn on "Show Far Side Markers" and set units to Right Ascension instead of degrees. Works nicely once you do.

closed 2019-01-22 by astrojonathan

pkgw, 2019-01-22:
Would there be a convenient way of making this Just Work™ without users needing to take those extra steps?

astrojonathan, 2019-01-22:
This is by design. We would not Want to change this behavior. When the data is in RA, all the values are also valid for Degrees, so we can't magically change it to degrees without the user decision. 

The far side markers are used for planetary mode, and have always been historically off by default, changing this would require existing users to change their behavior. 

We could default to differently for the far side markers when placed on the sky, or a large distance from the planetary body, but that is hard to explain, and harder to know ahead of time.

This also may change users who have scripted.

This issue was just that the author assumed there was a bug, instead of checking the settings.

pkgw, 2019-01-22:
I personally would still categorize this as a bug — most of our users are not going to be nearly as sophisticated as @astrodavid, and they are not going to read any manuals or GitHub discussions, so they're going to see the Local Group disappear as originally reported. It's certainly relevant that (1) the problem can be worked around with the right know-how and (2) an automagical solution is infeasible if not impossible, but the UX still has a shortcoming here.

astrojonathan, 2019-01-22:
If a used says their data is in Degrees and select degrees, and it is actually in RA hours. There is NOTHING we can do about that and that is not a bug. You have to actually tell the app the truth.
This is USER data not our data, the user had the responsibility of telling us how they are mapping this. I don't know how we can magically read there mind and pick a mapping that does not match what they asked?

astrodavid, 2019-01-22:
A little context from the data import, the hours/degrees label is next to declination instead of Right Ascension. Perhaps we could change the position of the units and add clarification that it is for RA units. For the record, I didn't understand why Dec would be in hours, but didn't process that those should be RA units instead.

I might have figured out the unit issue given a little more sleep (hard to come by lately), but I don't think that the "show far side markers" would be obvious enough for me to try. Maybe more clarification again in parentheses saying "select if using data at a large distance from the reference body" (with better wording)?

[image]

astrojonathan, 2019-01-22:
The image you attached has strange numbers in the Data Sources? Did you use a Column Header for your Column names? This is a huge help. All the existing mapping magic goes dark when you don't have a valid header.

astrojonathan, 2019-01-22:
We should create a new issue to track moving the UI for Units for both RA and Altiture down next to the Column mapping.

astrodavid, 2019-01-22:
Updated functional WTML file for anyone following who is interested.
[Local Group Galaxies.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/2784115/Local.Group.Galaxies.zip)