WorldWideTelescope/wwt-windows-client #131: Auto-hide Tabs causes image stretch

astrodavid, 2019-01-23:
First picture shows the image stretch with auto-hide tabs turned on, second picture shows a normal image with auto-hide tabs turned off.

The stretching also occurs when a tour is played and the menus hide.

[image]
[image]

astrodavid, 2019-01-23:
When Auto-hide tabs is enabled prior to WWT startup, the effect is backwards and turning off auto-hide tabs causes squishing inside.

Manual manipulation of menus also causes the stretching/squishing.

astrojonathan, 2019-01-23:
Does a screen resize fix this?

astrodavid, 2019-01-23:
Yes it does until the menus are opened or closed again.

astrodavid, 2019-01-23:
If it helps, I tried to save the screen image through WWT and both events are correctly saved through WWT, despite what the screen looks like.

astrojonathan, 2019-01-23:
It appears like the backing buffer is a different size than the screen it is drawing to.

philrosenfield, 2020-05-31:
Can this be closed?