WorldWideTelescope/wwt-windows-client #133: Honor MSI parameters for scripted installations

pgalbraith, 2019-02-25:
The MSI installer seems to use the parameter `AUTOUPDATE` to determine if autoupdates should be enabled, but passing `AUTOUPDATE=0` as a parameter to the installer gets ignored and `AUTOUPDATE` is changed from 0 -> 1 during the install.  It would be nice if the `AUTOUPDATE=0` value passed on the command line at install time would be honored by the installer.

Similarly for the legacy installer both `AUTOUPDATE=0` and `LAUNCHEXCELADDIN=0` are not honored from the command line.

pkgw, 2019-03-26:
CC @astrojonathan, @philrosenfield for the planned Windows release.

pkgw, 2022-01-05:
It's taken a while but I think I have a fix for this issue! It ended up being a real mess -- gritty details in #194. The candidate fix should make it into the next beta release, which I want to put out soon so that I can test the auto-updating functionality more generally.