WorldWideTelescope/wwt-windows-client #134: Integrate WWT with Urho

XuequanZhang, 2019-03-17:
Hello, WWT is a really good 3D platform, but the simulation ability is limited. I would like to integrate WWT with a light game engine, named Urho(https://github.com/xamarin/urho). While I get the problem how to render Urho scene and the virtual globe of WWT in the same control. How to integrate Urho as a basic rendering engine for WWT?
Any body have an idea? Thanks very much!

pkgw, 2019-03-26:
Can you please explain what you expected to see in the screenshot above? Thanks!

XuequanZhang, 2019-03-27:
Thank you very much for your reply. Since the polyline is not on the terrain (in the sky), when I rotate the camera, the polyline appears to be moving (not fixed to the ground). I want to fix the bug to render the vector data on the terrain just like an image.