WorldWideTelescope/wwt-windows-client #135: Detach view to Full Dome

fadetoben, 2019-06-07:
I work for ASH Enterprises. We have a bounce mirror Full-dome projection system called Warped Media. One complaint we have had from **all** of our clients who use WWT in Full-dome is that when you detach view to a second monitor, you no longer have a preview on your main screen. For our clients this means they have to try to be looking up on the dome while using their keyboard and mouse to do things. To me this seems like a glaring oversight. Therefore, my request is that when you detach view to a second monitor, you have the toggleable ability to have a preview on your first monitor.

astrojonathan, 2019-06-07:
You may want to use try the View->FullDome->"New Full Dome Instance" menu option and specify the screen. That will run a second copy of WWT in full dome mode, while leaving the master alone. This works more like WWT does in Multi-Channel projection mode.  We have both modes for peoples whos laptops can't handle rendering 6 times the polygons that is required to handle the full dome and a master at the same time.

pkgw, 2019-06-08:
I just wanted to chime in to mention that this question would be a great candidate for raising on our new WWT forum site:

https://wwt-forum.org/

We'd love for you to join and help get that community off the ground!

philrosenfield, 2020-05-31:
Can this be closed?

pkgw, 2020-06-23:
This sounds like a pretty popular feature request, this is worth keeping open.