WorldWideTelescope/wwt-windows-client #137: Installer installs to "Microsoft Research" subfolder of Program Files

pkgw, 2019-06-25:
For AAS branding this should change to "American Astronomical Society".

astrojonathan, 2019-06-25:
While branding is nice, this could cause compatibility problems for tools and installation that expect the location path to exist. The side effects of this could be devastating to KIOSK and dome installations that have automation supported. We did not change this in the past because of those concerns.

MrGrymReaper, 2020-01-09:
How about setting up a symlink of hardlink to the folder being used currently. Announcing that the old path ("Microsoft Research") is being deprecated, with enough time for these troubling installations to be altered. Once the time period has past a future installer could automatically remove that symlink/hardlink from the system.

The installer could start using the new path from today or one of the future releases, while also creating some form of symlink or hardlink. These forms of links are usable from Windows 7 or above which are the ones currently supported (on non-legacy releases). So it could create a symlink or hardlink called "Microsoft Research" (unless there's still a part of Microsoft Corporation), making releases under this name. Which would then point to the new location "American Astronomical Society" also in the "Program Files" folder.

philrosenfield, 2020-05-31:
Since at least Windows 7, I'm not sure if any brand benefits by holding the install folder title. If the change is approved, I'd suggest calling the folder WorldWide Telescope Programs or something similar.

pkgw, 2020-06-23:
My main concern is I don't want to be putting something out that gives the impression that it's a product of Microsoft Research when that's not the case.