WorldWideTelescope/wwt-windows-client #139: The moon is not tidally locked to Earth as it should be in Solar System Mode

astrojonathan, 2019-07-16:
The moon is not tidally locked to Earth in Solar System Mode as it should be. 
Go back to several dates, focus on the moon, zoom back, rotate view to put earth just in front of the moon. Zoom back into moon, notice the face is not the same at every date. It seems the tidal locking logic is not working in Either Web or Windows Client.

pkgw, 2020-08-10:
@astrojonathan Pat asked me to check in on this issue. I'm having trouble finding any code that seems to implement tidal locking. Is there supposed to be logic that specifically tweaks the moon's rotation to keep it locked toward the Earth? The moon's rotation period in Planets.cs is just specified as "27.3", which doesn't look precise enough to achieve locking just through the numerics.

This issue is potentially also relevant to part of #123.

patudom, 2021-12-06:
@pkgw, bumping this as an issue that ought to be fixed in the new release if possible.

pkgw, 2021-12-06:
Thanks for taking a look at the backlog! We'll have to see but hopefully @Carifio24 will get a chance to look at this. Note also the corresponding bug WorldWideTelescope/wwt-webgl-engine#61 for the web version.