WorldWideTelescope/wwt-windows-client #140: While loading csv files containing Chinese appears garbled

qu123xxx, 2019-07-18:
When I load a csv file in below ,WWT apperes garbled.

[image]

[image]

When I copy all the data in the file and paste into wwt. It back to nomal.
[image]

But I have hunders of files, and I want to use pywwt to create different groups of layers in WindowsClient. Then save it as .wwtl file. Thus I can't  do it manual. But,

1.In pywwt, the `new_layer()` method isn't support 'Geometry' like `TEXT()` or 'Linestring()'.
  The `load()` method happen to the same issue like above.

2.Excel-addin is unavailable.

It make me unable to create about 200 layers automatically.

qu123xxx, 2019-07-18:
I find the solution.When I use notepad open the csv file I find the `Linestring` and `TEXT`  is between `"`.
[image]
Delete it and save csv as Unicode or UTF-8 the error would gone.

pkgw, 2019-07-25:
Thanks for the update — do you think this issue can be closed now?

qu123xxx, 2019-07-26:
YES，I think it can be closed.

closed 2019-07-30 by pkgw