WorldWideTelescope/wwt-windows-client #142: Ability to sync tour with telescope motion

pkgw, 2019-08-05:
Reported by Aidan Cook, CfA summer intern:

"It would be great to add the ability to synchronize a guided tour with the slewing of the telescope so that the tour would not have to be exited to move the telescope. This would require a cap on slew speed of the tour to match the max slew speed of the telescope. It also would probably place an even higher performance demand on the computer/application."

astrojonathan, 2019-08-05:
This would not be practical. There are many types of telescopes from "Push to" to giant building sized mounts that take significant time and resources to slew. We can't prevent a tour (by design) from "slewing too fast". Besides we would not know what "too fast" was anyway since telescopes vary. 

We should consider other ways to design tours that would allow WWT to show the content, then allow the user to slew the telescope while the tour was paused, or waiting at a slide.

Also looking at a computer screen will ruin your night vision. Using WWT to move telescope while imaging is one thing, but looking at a screen, looking thru an eye piece and back again will not be an optimal task for human ocular biology.

philrosenfield, 2020-05-31:
> We should consider other ways to design tours that would allow WWT to show the content, then allow the user to slew the telescope while the tour was paused, or waiting at a slide.

+1 This would be a feature that would increase accessibility. Having a screen inside an open night/observatory can help share the experience with folks with low vision or not mobile enough to get to the eye piece.