WorldWideTelescope/wwt-windows-client #148: GLIMPSE/MIPSGAL corruption

marksubbarao, 2020-01-14:
When I try and load the GLIMPSE/MIPSGAL image it only loads the base tile layers. After that WWT won't load any tiles for me from any image, until I restart.

AstroRob, 2020-01-14:
Interesting; I was about to open an issue on the missing/empty GLIMPSE 360, but on my client (the current public 6.0 beta) the GLIMPSE and GLIMPSE/MIPSGAL images load correctly (from the all sky surveys collection). However I only tried it once right after installing (current Windows 10 on iMac Boot Camp). And G360 does not show up either on the main background pop up menu or from the collection.

pkgw, 2020-01-20:
I recall @astrojonathan seeing some issues along these lines that turned out to be the fault of the bad internet in the Honolulu convention center. Can you confirm if these issues persist when you have a good internet connection? @astrojonathan would they need to clear their caches or anything like that?

astrojonathan, 2020-01-28:
We have a CDN issue. Every request that routes Glimpse thru the CDN retreives the same image for 0,0,0.  http://worldwidetelescope.org/wwtweb/g360.aspx?q=4,2,2 shows one image

and http://cdn.worldwidetelescope.org/wwtweb/g360.aspx?q=4,2,2 shows just the root.

astrojonathan, 2020-01-28:
There was recursive FileDelete call that should have been a File.Delete call, this might have been an issue originally.

pkgw, 2020-02-19:
I am investigating this issue. It does seem that it's just an issue with the CDN having some bad data in it, although note that things are a bit hairy since you only get the bad data from some CDN point-of-presence nodes. (I.e., for Jonathan's example URL, I get a bad result (`No image`) at Harvard, but it is fine from home.)

I *think* that to fix this we just need to purge the cached data from the CDN nodes, but this is turning out to be a real pain since our CDN configuration seems to have something funky where we're not able to purge the cache in the way that one normally would. (This is probably because our CDN setup is quite old, in cloud terms.)

pkgw, 2020-02-20:
After a surprising amount of difficulty, I have gotten the CDN cache purged and my test case is now working. @marksubbarao are you able to confirm if things seem to be working better now?

marksubbarao, 2020-02-21:
@pkgw It does not seem to be any better for me.

AstroRob, 2020-02-21:
@pkgw<https://github.com/pkgw> @marksubbarao<https://github.com/marksubbarao>  I should clarify that I had checked this in the web client forgetting it was a desktop issue.

I tried it again in the desktop client and only saw a very low res preview image. I then purged the image cache and restarted WWT and now the GLIMPSE360 is completely blank in either the background selector or the collection.

GLIMPSE and GLIMPSE-MIPSGAL continue to work as expected.

  Robert Hurt, Ph.D.       |   626-395-1825 (office)
  Visualization Scientist  |   (ICE) IPAC Communications & Education
  Caltech/IPAC             |   NASA’s Universe of Learning
------------------------------------------------------------------------
  UniverseUnplugged.org<http://UniverseUnplugged.org> -  astropix.ipac.caltech.edu<http://astropix.ipac.caltech.edu>
------------------------------------------------------------------------

pkgw, 2020-02-26:
OK, I think we need to change the settings of the CDN infrastructure to treat URLs with different query strings (`?q=N,X,Y`) as different and cache them individually. This is supported, but I need to go through the CDN provider to get the configuration set up. Working on it.

pkgw, 2020-03-13:
I am told that the CDN settings have been updated. If you purge your cache, does GLIMPSE360 seem to be working now?

pkgw, 2020-04-01:
Closing this issue because I believe this is working now. Please reopen if that's incorrect!

closed 2020-04-01 by pkgw