WorldWideTelescope/wwt-windows-client #149: Why the same RGB HEX code result in different color in wwt windows client and other place?

qu123xxx, 2020-02-24:
Hi I have encounter a problem when I try to visualize a color through hex code of **RGB** `：#103D25`

If I input this code in any color translate website, I can see this color: 
[image]

If I input it in wwt windows client with **RGBA hex code** `FF103D25` I will see this color:
[image]

**I use RGBA hex code is because I can't display any color with RGB hex code.**

So I want to ask, what's happen to the color and how to display correct color in wwt windows client?

astrojonathan, 2020-02-24:
The issue is how applications handle GAMMA. Color representation in 3D Computer graphics vs. web run thru very different pipelines. Web Colors have sRGB color representation that has a GAMMA baked into it. in 3D Graphics with Alpha, Colors are linear so when they are anti-aliased or blended they show up correctly, then the entire output is GAMMA corrected for display. This means that colors can vary. You can try to apply a de-gamma curve to the RGB values and use that hex code if you want an exact visual match.

qu123xxx, 2020-02-24:
Thanks for your reply～ Let me try tomorrow morning～

qu123xxx, 2020-02-25:
I find the GAMMA Correction is a bit difficult.[
I use the ](https://www.colour-science.org/#)color-science module in python to adjust the color:

Original color RGB values from matplotlib is [0.16813,0.45999,0.55808], hex code is #2a758e
[image]

```
import colour
import numpy as np

RGB_Mpl=np.array([0.16813,0.45999,0.55808])
RGB_WWT=colour.gamma_function(RGB_Mpl,exponent=2.1)
WWT_hex=colour.notation.RGB_to_HEX(RGB_WWT)
```

WWT_hex is #06314a
[image]

But they are still have some differences. 
**How can I get the right RGB values to get exact visual match ? Is there any fixed GAMMA numbers to adjust it ?**

pkgw, 2020-03-20:
Because web and 3D color models and processing pipelines are different, it may not be feasible to obtain an exact match across an entire palette of colors. Assuming and un-applying a gamma of 2.1 is probably the best that you're going to be able to do without a putting in a *lot* more effort.

qu123xxx, 2020-03-21:
Thanks for you reply~ I think I have found a way to solve it. 
I try to use the screen capture script to get the RGB in a specific place in WWT and adjust the gamma automatically to make the input_data same as the capture_data. I think it would be fast.

closed 2020-04-01 by qu123xxx

qu123xxx, 2020-04-23:
[image]

I have done it.

pkgw, 2020-04-23:
@qu123xxx Cool! If you'd like, a post on the [WWT Forum](https://wwt-forum.org/) documenting what you did would be a great way to preserve the workflow and potentially help out anyone else in the future trying to achieve the same effect.