WorldWideTelescope/wwt-windows-client #150: Possible caching bug in UWP build

pkgw, 2020-03-24:
I've been checking out the Windows client codebase to understand the web calls that it makes since it would be pretty bad to break those as I work on the web infrastructure! I noticed something that worries me.

When the Windows client logs in, it does the update checks and gets the data version from the server. If the data version has changed, it sets `DataSetManager.DataFresh = False`.

From what I see, that variable is only used in in `DataSetManager.DownloadFile()`, so that seems to be the core routine where the Windows client decides that it needs to refresh its data.

But the DataFresh flag is only referenced on the `#if !WINDOWS_UWP` branch of the code (line 149). On the `WINDOWS_UWP` side, it seems to unconditionally download the file, with a try/catch that looks like it will fire if the file exists. The DataFresh flag isn't referenced.

So my worry is that, if I'm understanding the pieces correctly, UWP clients won't update their data caches when told to by the server, because as far as I can tell that information is only propagated through the DataFresh flag.

astrojonathan, 2020-04-02:
UWP client does not do much caching, it acts more like the web client and downloads files ignoring fresh state. We don't ship any "built" projects yet on UWP, it is experimental in nature right now, so its not impacting end users.

pkgw, 2020-04-03:
OK that is good to know. I had misunderstood the broad context and had thought that the 6.0 Windows client was UWP based, so I was worried that this issue would have a much bigger impact.

philrosenfield, 2020-05-31:
Can this be closed?

pkgw, 2020-06-23:
Looks like there's a potential issue here, but unlikely to cause problems in practice — I'll go ahead and close it.

closed 2020-06-23 by pkgw