WorldWideTelescope/wwt-windows-client #152: 6.0 beta: constellations appear at radius of 1 AU in Solar System mode

pkgw, 2020-04-01:
Reported by @astrodavid [on the forum](https://wwt-forum.org/t/constellations-in-ss-mode-windows-client-6-0-11-1/104). The screenshot shows it all:

[image]

AstroRob, 2020-04-01:
Unhelpful comment: Were that sphere centered on the Earth rather than the Sun I think I would rank this as a “Feature” as it would emphasize how constellations are merely a human projection onto a changing field of stars that has no interest in the groupings humans assign to them… Symbolically having it localized to the Earth would establish it more artfully than any collection of words could manage...

[ConstWWT]<https://user-images.githubusercontent.com/59598/78141895-2485fb00-73fa-11ea-9399-f48161fe4379.jpg>

------------------------------------------------------------------------
  Robert Hurt, Ph.D.       |   626-395-1825 (office)
  Visualization Scientist  |   (ICE) IPAC Communications & Education
  Caltech/IPAC             |   NASA’s Universe of Learning
------------------------------------------------------------------------
  UniverseUnplugged.org<http://UniverseUnplugged.org> -  astropix.ipac.caltech.edu<http://astropix.ipac.caltech.edu>
------------------------------------------------------------------------

astrojonathan, 2020-04-01:
Oops. It appears that the DrawImage is reverting coordinates back, so after the first constellation image is drawn, the rest are drawn at 1AU. Needs fixing obviously, but I know where and why now.

astrojonathan, 2020-04-02:
DrawTiledSphere was restoring the worldbase and world to a copy of worldbase that was saved. I split it to save both independently and this fixes the issue. Will PR this with the Beta Update.

closed 2020-04-02 by astrojonathan