WorldWideTelescope/wwt-windows-client #153: Text Opacity Doesn't Work As Expected When Text is Animated

thallstd, 2020-04-29:
I’ve tried using opacity to fade text in and out but can’t get it to work. I’ve set the text to Animate but if I set the start camera position to 0 opacity and the end to 100 then run the tour the text is visible for the duration of the slide.  If I set them in the other order and run the tour the text is not visible during the slide. 

2 others in the WWT forum have confirmed the same behavior and consider it a bug.

Here is a screencast of the issue and the steps to reproduce it: https://www.loom.com/share/3c14ca94c66e4677ae702a3eeec2eca6

1) On a new tour add a slide and add text to it. You can leave the slide black.
2) Check the Animate option for the text
3) Go to Start camera Position and change text opacity to 0 (it will vanish)
4) Go to End camera position and text is no longer visible. It too is vanished - even though the opacity still shows as 100.  
5) Save it as 100.
6) Go back to Start camera position.
7) It is now visible there as well.  Resv opacity as 0.
8) Play the tour.

No matter which camera position is used to set the opacity, the other camera position inherits it when the tour is played.

astrojonathan, 2020-04-29:
This is definitely a bug. I was able to repro it myself. What release of WWT are you running?

patudom, 2020-04-29:
@astrojonathan, I tried reverting my WWT install to older versions (6.0.04, 5.5.03, 5.2.9, 4.1.74), and this bug is in all of them. I then found a note from 10/2012 that I made about this issue, but it never got carried over to the github issues list.

astrojonathan, 2020-04-30:
Thanks. That is very useful. It helps track down what is going on! It's not just a issue introduced with the AAS Jan meeting release.