WorldWideTelescope/wwt-windows-client #154: Error occur when I input data through pywwt

qu123xxx, 2020-05-08:
When I try to pass a 322322 rows table  to WWT windows client through `layer.update` in `pywwt`, I got this error in visual studio 2017 debug mode

`Error: System.ArgumentOutOfRangeException: ?????????????? DateTime? ???: t    ? System.DateTime.op_Subtraction(DateTime d, TimeSpan t)    ? TerraViewer.SpreadSheetLayer.PurgeByTime()    ? TerraViewer.SpreadSheetLayer.LoadFromString(String data, Boolean isUpdate, Boolean purgeOld, Boolean purgeAll, Boolean hasHeader)    ? TerraViewer.SpreadSheetLayer.UpadteData(Object data, Boolean purgeOld, Boolean purgeAll, Boolean hasHeader)    ? TerraViewer.LayerManager.UpdateLayer(Guid layerID, Object data, Boolean show, String name, Boolean noPurge, Boolean purgeAll, Boolean hasHeader)    ? TerraViewer.HTTPLayerApi.Proc`

And the python.exe terminal show this messae:
```
D:\Tools\Visual Studio 2017\Shared\Python36_64\lib\site-packages\pywwt\windows\misc.py:24: UserWarning: No parser was explicitly specified, so I'm using the best available HTML parser for this system ("lxml"). This usually isn't a problem, but if you run this code on another system, or in a different virtual environment, it may use a different parser and behave differently.

The code that caused this warning is on line 24 of the file D:\Tools\Visual Studio 2017\Shared\Python36_64\lib\site-packages\pywwt\windows\misc.py. To get rid of this warning, pass the additional argument 'features="lxml"' to the BeautifulSoup constructor.
  soup2 = BeautifulSoup(resp_str)
```

I try to google the first error. It seems that's from WWT windows client.

the data snapshot is here:
[image]

WWT version is 6.0 Beta
OS: windows 10

Thanks in advance for any replies.

astrojonathan, 2020-05-08:
Is the build debug or release? In debug it might be targeting Win32 and have a 4gb address limit. There may be a max string size and the error is cause trying to parse the last line cut in half. 

Could I try to repro this and see what the failure is?

qu123xxx, 2020-05-08:
what's the meaning of 

> Is the build debug or release?

I debug my code in visual studio 2017 debug mode.

Because my output file is nearly 300MB, I think you can download the raw data here

> http://www.astroblend.com/data/sphdec.tar.gz

and then merge middle 3 columns in first 25 `.txt files ` .
 the `layer` is created by `wwt.ney_layer()`.

layer's `frame = Sandbox`
 `CoordinatesType` in `property_dict` is `Rectangular`

Then pass the data to wwt layer use `layer.update`. The error1 may occur.
The error1 may need to use some IDE to reproduce.
Feel free to tell me if it can't be reproduced, I can send my file to you through email or other ways.