WorldWideTelescope/wwt-windows-client #168: Tour audio / music inconsistent start time

astrodavid10, 2020-09-07:
Audio in a tour does not always start playing at the same time (a bigger problem if trying to time time visuals precisely to audio).

pkgw, 2020-09-08:
Windows? Web? Both? Always, or only sometimes?

astrodavid10, 2020-09-11:
This is something that I have only noticed in the Windows client, but haven't really tested on the web. I have noticed it in the current build and previous versions as well. It occurs typically after the first time the tour is played after loading it. 

Perhaps related, if a tour is previewed from a specific slide instead of pressing play, audio sometimes does not play at all.

pkgw, 2020-09-11:
Thanks, David. When this happens, how big of an effect are we talking — fraction of a second? Multiple seconds? Hopefully not more than that ...

patudom, 2020-09-11:
@astrodavid Does it make a difference if you are using slides in Solar System mode vs. Sky mode?  @astrojonathan has helped me troubleshoot audio alignment issues with Solar System mode tours in the past, but maybe this would apply to Sky mode tours also. Basically, if I didn't have the start camera position of a slide line up perfectly in time with the end camera position of the previous slide (for example, if a few seconds had elapsed between when I generated the next slide), WWT would try to account for the time glitch in the audio. The alignment would be fine if I ran the tour from before the time glitch, but if I started the tour after the time glitch, the audio would start in a different place.  It would get worse and worse as more time glitches stacked up over multiple slides. (I have never used the timeline editor, so I don't know how that would interact with all this).

astrojonathan, 2020-09-11:
I believe I have figured out what is going on, but I need to verify. There is a "Cost" function that calculates how long a slide takes to run. The music is timed to start or jump based on the accumulated run time estimate. If the estimate is off, then this might not start at the correct place. I need to carefully look at the functions   CalculateRunTime() and ElapsedTimeTillTourstop in tour.cs and see if the computation is correct.

Here is where those functions live.
https://github.com/WorldWideTelescope/wwt-windows-client/blob/4a01be66438d841f836688186801610733327bde/WWTExplorer3d/Tour.cs#L1083

astrodavid10, 2020-09-14:
@pkgw I'd say its <~0.25 sec, but I also don't think it is always the same.

@patudom I think it applies to both modes, though in the most recent case I am in SS mode. I definitely do not have the end of the previous slides lining up with the start of the subsequent slide as I am jumping between reference frames. Each slide makes use of the timeline.

@astrojonathan I am looking forward to the fix :) - no pressure haha