WorldWideTelescope/wwt-windows-client #169: 3D Models flipped along one axis (Y?)

astrodavid10, 2020-09-11:
3D models are flipped long what appears to be the Y axis in both the Windows and web clients. Compare the attached example of Cassini with what it should look like: https://nasa3d.arc.nasa.gov/detail/jpl-vtad-cassini 
[Cassini.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/5205678/Cassini.zip)

Carifio24, 2022-02-25:
@astrodavid10 Yeah, it looks like the handedness of the models is flipped between the linked image and WWT.

Just curious, how did you make this WWTL file? The Cassini link gives a Blender file in the zip file. Apparently Blender uses a right-handed coordinate system, while DirectX uses a left-handed system, so I'm wondering if this is the cause of the discrepancy here.

There is a `FlipHandedness` parameter in WWT's `Object3d` class which might be able to help with this problem. I'm not sure where to import the 3d object file into WWT, though.

astrodavid10, 2022-03-02:
Hey, awesome! I appreciate it!

pkgw, 2022-03-09:
OK, I finally got around to finishing up the test build of this.

If you download [this "drop-signed" Zip file][x] and run the installer at `drop-signed → Setup1 → Release → WWTSetup.msi`, you should be able to test out the latest build that includes this fix.

[x]: https://artprodcus3.artifacts.visualstudio.com/Abc6fd740-4afc-4f26-9e0f-c9062c3aa36e/5b8413f0-309f-4655-933b-c3b9516cd60f/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2Fhc3dvcmxkd2lkZXRlbGVzY29wZS9wcm9qZWN0SWQvNWI4NDEzZjAtMzA5Zi00NjU1LTkzM2ItYzNiOTUxNmNkNjBmL2J1aWxkSWQvMjUxNS9hcnRpZmFjdE5hbWUvZHJvcC1zaWduZWQ1/content?format=zip