WorldWideTelescope/wwt-windows-client #170: Ref Frame as a Child of a Created Ref Frame Centers on the Sun

astrodavid10, 2020-10-20:
If I create a reference frame for an asteroid like Bennu and then create a ref frame as a child to show OSIRIS-REx's orbit at Bennu, it centers and scales for the sun instead. See the attached.
[OSIRIS-REx.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/5406409/OSIRIS-REx.zip)

pkgw, 2020-10-21:
Do you think this is a new issue? This seems like the kind of functionality that would have been exercised in the past ..

astrodavid10, 2020-10-21:
I don't think that it is new... I'm not sure I had ever put forth good effort to get a new ref frame to have children ref frames with something like a trajectory.

pkgw, 2021-12-03:
Flagged by @astrodavid10 as something to look at for the upcoming release. CC @Carifio24.

Carifio24, 2022-02-02:
@astrodavid10 I was looking into this using the example that you gave, but I'm unsure what I should expect to see at these reference frames. If I track Bennu, I see this:
[image]

When I track OSIRIS-REx, I see:
[image]

I can pan around and see the trajectory intersecting itself, like this:
[image]

I'm just curious what I should expect to see at the OSIRIS reference frame.

astrodavid10, 2022-02-02:
The green are the orbits of OSIRIS-REx around Bennu and they should be attached to Bennu and not the Sun. If you are tracking OSIRIS-REx and zoom out you'll notice that it is residing in the Sun.

astrodavid10, 2022-02-02:
@Carifio24 They are the correct shape and scale (or at least close) but are attached as a child of the sun incorrectly.

Carifio24, 2022-02-02:
@astrodavid10 Ah, ok. For some reason, I'm actually unable to zoom in or out when tracking either of these layers. I also get the warning that "This file does not seem to be a valid collection", but the layers still appear in the layer manager.

I know that parts of WWTL files are often compressed, so I'm wondering if decompressing the .zip that you attached had some adverse effect. Could you share the (unzipped) layer file?