WorldWideTelescope/wwt-windows-client #172: "Show End Slide" gives two different results when clicked back to back

astrodavid10, 2020-12-27:
Using animation for a picture, while moving the background sky and advancing time (viewing from a location in Sky mode), seems to cause the end slide to recognize two separate ending points. Clicking "Show End Slide" once provides one, and a second time, the other. I *think* that this may have to do with advancing time while moving the sky. Example tour attached.https://guides.github.com/features/mastering-markdown/
[animation_orion.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/5744164/animation_orion.zip)

astrodavid10, 2020-12-27:
Not using animations while time advances seems to bypass the issue.

pkgw, 2020-12-28:
Thanks for the report! I'm guessing that maybe there is a bug in how the frame progression state is synchronized between the main cameras and the animated items. For what it's worth, I suspect that the actual tour file should reliably play back as intended ... this is probably just an issue with how things are displayed in the editor, not the actual tour content.

astrodavid10, 2020-12-30:
Yes, with further testing, I can confirm that the tour plays as intended, despite confusion within the editor.