WorldWideTelescope/wwt-windows-client #178: VOTable radius/distance/etc columns not working

pkgw, 2021-12-03:
As reported by @astrodavid10: if you have a VOTable with 3D data, the distance axis doesn't seem to be working. Everything shows up at R = 1 AU.

Here's a gzipped WWTL sample file that demonstrates the problem: [exoplanets_vo_nov_2021.wwtl.gz](https://github.com/WorldWideTelescope/wwt-windows-client/files/7650912/exoplanets_vo_nov_2021.wwtl.gz)

> a VO table of exoplanets that simply extends to a shell in the SS mode in the Windows client.

Carifio24, 2021-12-06:
So I've identified where in the code for VOTable layers this is, and I _think_ I have it working, but it's a bit difficult to test since I don't know what the layer in this example should look like. @pkgw @astrodavid10 do either of you know what this layer should look like? The screenshots below are how I've got it displaying now. I'm a bit concerned about the first screenshot - I don't know a lot about the distribution of known exoplanet locations, but it seemed odd to me that some would be so far off of the galactic plane.

[image]
[image]

astrodavid10, 2021-12-06:
Hmmm, I'm not seeing the Kepler cone like I'd expect:

[image]
[image]

Carifio24, 2021-12-06:
@astrodavid10 Thanks for the screenshots, and the mention of the Kepler cone - I found what I think was the issue and am now getting the images below from this layer. Looks like the Kepler cone on the left and exoplanets found via microlensing going to the right.
[image]
[image]

astrodavid10, 2021-12-06:
Looking much better!

astrodavid10, 2021-12-07:
[PS_2021.11.30_19.50.39.xml.zip](https://github.com/WorldWideTelescope/wwt-windows-client/files/7665051/PS_2021.11.30_19.50.39.xml.zip)
Hmmm, should be parsecs. The initial votable, zipped and attached, can be added as a votable  from Explore -> Open -> VOTable and this does not auto recognize the distance parameter. When sy_dist is selected as the distance parameter, the units cannot be specified and so I guess it is not interpreting them correctly.

Carifio24, 2021-12-07:
@astrodavid10 Hmm, yeah, it seems that the VOTable viewer just [assumes that the distance unit is light years](https://github.com/WorldWideTelescope/wwt-windows-client/blob/master/WWTExplorer3d/VOTableViewer.cs#L366). It looks like the VoTable already stores the unit (as a string) for each column, so it should hopefully be a quick fix.

astrodavid10, 2021-12-14:
@Carifio24  @pkgw  Initial testing looks very good to me, thanks! As I pan around the exoplanet data while zoomed out, individual points end up fading (well, it looks like clipping and it is abrupt) in and out. Is there any way to control this such that the whole dataset can be viewed without any part being removed?

pkgw, 2021-12-14:
@astrodavid10 This sounds like it could be the "far side visible" setting that I mentioned in the other issue. Do you happen to know how to control that setting in the UI? I think there's almost definitely a way, but I only know how to get at it from Python.

astrodavid10, 2021-12-14:
@pkgw  That is what I figured, but as far as I can tell, there is no way to control that parameter through the VOTable UI. The properties button is not functional for VOTables and the VOTable Viewer doesn't have that setting.

pkgw, 2021-12-14:
@astrodavid10 Thanks for reporting back. Either Jon or I can take a look through the source code to see if we're missing anything ... and also maybe try a hack to see if the setting is actually relevant anyway.